wouterverweirder/kinect2 #3: 3D Joint Coordinates - Hand States

SpencerIO, 2015-01-05:
Hi Wouter,

node-kinect2 works great. Really enjoying having access to the skeleton in javascript. Right now I get the x and y coordinates for the joints. What would it take to implement the z coordinates for the joints as well as the hand (open, close, and finger) states?

wouterverweirder, 2015-01-05:
The hand states should be in the body data (handLeftState and handRightState).
Will investigate on adding alternative coordinate spaces (3D, depth, rgb) to the data.

David-dp-, 2015-04-06:
I agree with SpencerIO -- this is a really useful project. And I would also like access to depth values (and potentially everything available in the Body class https://msdn.microsoft.com/en-us/library/microsoft.kinect.body.aspx)

I had a look through the C# source, and it appears that a good way to allow users to configure a run is for them to pass a prototype response object into Invoke(input) of NodeKinect2.cs. It might also be a good idea to not hard-code coordinate mapping in BodyReader_FrameArrived but instead allow it to be passed-in as part of the configuration, maybe as a post-processing callback.

wouterverweirder, 2015-07-30:
I've added the extra coordinate spaces and hand states to the body frame info - enjoy!

closed 2015-07-30 by wouterverweirder