wouterverweirder/kinect2 #5: color feed browser -node server crashes on launch

jBachalo, 2015-03-26:
Hi Wouter
I have a perplexing issue. The color feed browser node application seems to crash shortly(less than a second) after launch. I even went back to an earlier version with no zlib. In the index.js file if I comment out these lines, the node server runs and does not crash.

resizedBuffer[j+1] = data[i+1];
resizedBuffer[j+2] = data[i+2];
resizedBuffer[j+3] = data[i+3];

What is even more perplexing is this seems to be machine specific. I was able to get the app running properly on my laptop running Windows 8.1, but NOT another machine. Can you think of anything hardware, or OS specific that might be the issue?  I'm fairly certain the nodejs node_modules are correct as I simply did a npm install in the root and colorfeed browser directories. Am getting no errors when I run, it simply quits less than a second after starting.  Any help appreciated!

jBachalo, 2015-03-27:
Resolved by upgrading edge module.

closed 2015-03-27 by jBachalo

wouterverweirder, 2015-03-27:
Cool, thanks for looking into this and posting how you resolved it!