wouterverweirder/kinect2 #16: Compilation/Installation issues

craiglonsdale, 2015-08-24:
Hey team,
I'm having a bit of an issue installing this Kinect2 module. It looks like the `GetKinectDefaultKinectSensor` can't link in the C++ part of this install.
I have had a look at the VS project to try figure out what the issue might be but  I can't see any real issues.
This is my stack 
```
if not defined npm_config_node_gyp (node "c:\Program Files (x86)\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" configure build )  else (node  configure build )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  kinect2.cc
     Creating library c:\Meerkats\Projects\kinect\node_modules\kinect2\build\Release\kinect2.lib and object c:\Meerkats\Projects\kinect\node_modules\kinec
  t2\build\Release\kinect2.exp
kinect2.obj : error LNK2001: unresolved external symbol _GetDefaultKinectSensor@4 [c:\Meerkats\Projects\kinect\node_modules\kinect2\build\kinect2.vcxproj]
c:\Meerkats\Projects\kinect\node_modules\kinect2\build\Release\kinect2.node : fatal error LNK1120: 1 unresolved externals [c:\Meerkats\Projects\kinect\nod
e_modules\kinect2\build\kinect2.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\12.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (c:\Program Files (x86)\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Windows_NT 6.3.9600
gyp ERR! command "node" "c:\\Program Files (x86)\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "build"
gyp ERR! cwd c:\Meerkats\Projects\kinect\node_modules\kinect2
gyp ERR! node -v v0.12.7
gyp ERR! node-gyp -v v2.0.1
gyp ERR! not ok
```

craiglonsdale, 2015-08-24:
Well after a bit more looking I found it :|
I changed
```
{
  "targets": [
    {
      "target_name": "kinect2",
      "sources": [ "src/kinect2.cc" ],
      "include_dirs": [
        "<!(node -e \"require('nan')\")",
        "$(KINECTSDK20_DIR)\\inc"
      ],
      "libraries": [
        "-l$(KINECTSDK20_DIR)\\lib\\x64\\kinect20.lib"
      ]
    }
  ]
}
```
to 
```
{
  "targets": [
    {
      "target_name": "kinect2",
      "sources": [ "src/kinect2.cc" ],
      "include_dirs": [
        "<!(node -e \"require('nan')\")",
        "$(KINECTSDK20_DIR)\\inc"
      ],
      "libraries": [
        "-l$(KINECTSDK20_DIR)\\lib\\x86\\kinect20.lib"
      ]
    }
  ]
}
```

Seems the x64 library might no have the same method signature?

wouterverweirder, 2015-08-24:
Thanks for the info - library is setup for 64-bit. Will look into adding 32-bit support.

wouterverweirder, 2015-08-24:
Might need to add a compiler condition, as seen in https://github.com/nodejs/node-gyp/wiki/Linking-to-OpenSSL

HectorArellano, 2015-08-25:
Hello! I'm having the same issue, but I don't know exactly what and where I should change. Any kind of light for this would be really appreciated!

HectorArellano, 2015-08-25:
Solved it, and it's working like charm! thanks!

craiglonsdale, 2015-08-26:
It seems that the reason my 64bit OS was using the 32bit library is possibly because when I'm running the npm install from my cygwin console, the architecture is `ia32`.

wouterverweirder, 2015-08-30:
ok, I've added a condition to the binding.gyp file for 32-bit versions - could you try with the latest version (0.0.30) and see what happens?

HectorArellano, 2015-09-01:
For me is working very nicely. I just wish it could have the orientation of the joints too. Sadly it only has the camera positions of the points. Anyway I guess I can reconstruct the orientation based on the relationship of the joints.

Fantastic library, kind regards for all the help.

newmovieclip, 2015-09-01:
I was working on that the other day. Stay tuned.

closed 2015-09-01 by wouterverweirder