wouterverweirder/kinect2 #19: socket.io ERR_CONNECTION_REFUSED

mroglic, 2015-09-02:
When I run any example and go to localhost:8000 a get a blank screen and this error in console:

GET http://localhost/socket.io/?EIO=3&transport=polling&t=1441152584189-0 net::ERR_CONNECTION_REFUSED

I assume that socket.io path could be wrong. Here is the structure of my folders:

C:\Users\mr\Desktop\node_modules
express
kinect2
socket.io

And then I run it in this folder C:\Users\mr\Desktop\node_modules\kinect2\examples\color-feed-browser>node index.js
Server listening on port 8000
Point your browser to http://localhost:8000

wouterverweirder, 2015-09-02:
This might be windows firewall preventing socket setup. Do basic socket.io applications work? Have you tried other ports (instead of 8000)?

mroglic, 2015-09-08:
It's working now, the problem was that I didn't install express and socket.io correctrly. Thanks!

closed 2015-09-10 by mroglic