wouterverweirder/kinect2 #21: Electron : The specified procedure could not be found

activetheory, 2015-11-05:
Hi,

Thanks so much for this awesome lib.

I am trying to get it working using electron, and having some trouble. I have compiled based on the script in the readme, but am getting the following error when I run "electron ." in any of the example directories.

`Uncaught Error: The specified procedure could not be found.
E:\Code\Kinect\node_modules\kinect2\build\Release\kinect2.node`

I am certain the file exists, and here is the output of the build script
[image]

Thanks in advance.

RomainKurtz, 2015-11-05:
Hi,
I have exactly the same problem.
(And many thank you too for this wonderful lib.)

wouterverweirder, 2015-11-05:
Will investigate tomorrow (away from keyboard today). Could you check if this has something to do with electron versions? Could you try compiling & running with electron 0.30.4 (build script default).

Also: could you try running the example from the kinect2 dir? (electron examples/etc...) instead of inside that example dir.

activetheory, 2015-11-05:
Thanks for the tip, I ended up compiling with `node tools\electronbuild.js --target=0.34.0 --arch=x64` and that worked!

closed 2015-11-06 by wouterverweirder

caitoulin, 2019-03-21:
How to solve it ?  I have the same question.

commit 614c78e1 (2015-11-06): updated electron build to default to installed version number - closes #21