wouterverweirder/kinect2 #26: How can I obtain distance of an object using rawDepthFrame?

saurabh21289, 2016-03-10:


SujeethJinesh, 2016-11-12:
I would like to know this as well, have you found a solution?

saurabh21289, 2016-11-13:
Yeah I did. You can use the jointPoint.depthX, jointPoint.depthY, jointPoint.depthZ methods. They return distances in meters taking the Kinect as the origin.

SujeethJinesh, 2016-11-14:
@saurabh21289 If you were to do it generally (for example find the depth of the closest pixel > 50 cm away) how would you do that using RawDepthFrame? Thanks for your quick response by the way!

wouterverweirder, 2016-11-14:
You can loop through the values of the depth frame and check which is the smalles value. Something like this (based on https://github.com/wouterverweirder/kinect2/blob/master/examples/point-cloud-electron/index.html)

```javascript
var smallestDepth = 9999999;
var smallestX = 0;
var smallestY = 0;
for(var i = 0; i < depthBuffer.length; i+=2) {
   var depth = (depthBuffer[i+1] << 8) + depthBuffer[i]; //get uint16 data from buffer
   var x = i % 512;
   var y = Math.floor(i / 512);
  if(depth < smallestDepth && depth > 500) {
     smallestDepth = depth;
     smallestX = x;
     smallestY = y;
  }
}
```javascript

SujeethJinesh, 2016-11-14:
Thanks so much!

closed 2016-11-15 by wouterverweirder