wouterverweirder/kinect2 #28: Body.trackingId not unique

carstenschwede, 2016-04-07:
Once in a while I observe that a single BodyFrame contains multiple bodies with the same trackingId.

    kinect.openBodyReader();
    kinect.on("bodyFrame", (rawData) => {
    	var seenBodyIds = {};
    	rawData.bodies.map((body) => {
    		if (seenBodyIds[body.trackingId]) {
    			console.warn("Should not have duplicate tracking id",body.trackingId);
    		}
    		seenBodyIds[body.trackingId] = true; 
    	});
    });
    // $> Should not have duplicate tracking id 72.057.594.038.335.940

The bodies themselves were not duplicates, i.e. both included unique joint tracking data.

iTrackingId as an UINT64 has a max value 18.446.744.073.709.551.615: 

    UINT64 iTrackingId = 0;
    hr = pBody->get_TrackingId(&iTrackingId);
    m_jsBodyFrame.bodies[i].trackingId = iTrackingId;

JavaScripts Number.MAX_SAFE_INTEGER is 9.007.199.254.740.991:

    Nan::Set(v8body, Nan::New<v8::String>("trackingId").ToLocalChecked(), Nan::New<v8::Number>(static_cast<double>(m_jsBodyFrameV8.bodies[i].trackingId)));

The last duplicate trackingId according to my log was 72.057.594.038.335.940 (> Number.MAX_SAFE_INTEGER).
In case this is a problem with the casting process/max value of datatype, switching trackingId from v8::Number to to v8::String might help.

wouterverweirder, 2016-04-07:
Thanks for the detailed report. Will look into this on my windows machine :-)

carstenschwede, 2016-04-14:
After changing body.trackingId to v8::String I haven't encountered any duplicates so far.

commit b0f8f0e3 (2016-04-14): use javascript string to store the tracking id - fixes #28
closed 2016-04-14 by wouterverweirder