wouterverweirder/kinect2 #29: can't run browser examples

amejias101, 2016-04-11:
I can run the Electron examples just fine but I keep getting the following error when I try to run any of the browser examples:

```
C:\Users\alex\Desktop\kinect2-master>node examples\color-feed-browser\index.js
module.js:440
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: The specified module could not be found.
\\?\C:\Users\alex\Desktop\kinect2-master\build\Release\kinect2.node
```
I tried doing `npm install` in any of the browser examples, followed by `node index.js` but the above error keeps happening. 

```
C:\Users\alex\Desktop\kinect2-master>node -v
v5.10.1

C:\Users\alex\Desktop\kinect2-master>npm -v
3.8.3

C:\Users\alex\Desktop\kinect2-master>node-gyp -v
v3.3.1
```

wouterverweirder, 2016-04-13:
Might have something to do with your node version. There's no precompiled version of my library yet for 5.x - so it falls back to a locally compiled version, which might be the electron version in your case? Could you try re-installing my module (without the electron build)?

amejias101, 2016-04-13:
since I'm running the examples off this git repo, what exactly do you mean by reinstalling?

wouterverweirder, 2016-04-14:
Allright. Try running deleting the node_modules folder and running npm install again in the root directory of this repo. This should replace your electron version of this module back with the node version.
I've also created a precompiled version for node 5.x in the latest release of this library.

amejias101, 2016-04-14:
I forgot to update you yesterday but the browser examples worked with node v4.x.x. 

I just tried again with node v5.x.x but now it complains about about the the kinect2.node module missing. I tried running the snippet in the readme and changed the electron version to 37.5, it still works in electron but fails with a missing module in the browser.

Side note, how exactly should I be running the browser examples? right now I'm doing an `npm install` from the example directory, followed by `node .\examples\color-feed-browser`(or any browser example)

amejias101, 2016-04-14:
just realized that if I run `$ electron .\examples\[browser-example]` it works. Not sure why I need electron to run a browser example though

wouterverweirder, 2016-04-15:
That's because you've got a compiled version of this module for electron.
Try removing both the "build" folder and the "node_modules" folder and run npm install again.

closed 2016-04-19 by wouterverweirder