wouterverweirder/kinect2 #30: Hand tracking in the browser

Loyth, 2016-04-13:
Hello, i'm French so excuse me for my bad English.
I have a problem with hand tracking.

Indeed, I have implemented this code : 
```
var x = jointPoint.depthX * innerWidth; // 1920
var y = jointPoint.depthY * innerHeight; // 1080

hand.css('left', x);
hand.css('top', y);
```
But when my hand move in the browser, i can't reach the corners.

So, Do you have a solution to increase the sensitivity of the hand?

Thanks you.

carstenschwede, 2016-04-14:
depthX and depthY are going to be in meters, so probably for use cases in the range of [-0.5m,0.5m].
Something like
    
    var maxPixelWidth = window.innerWidth;
    var minRangeX = -0.5; var maxRangeX = 0.5;
    var x = (jointPoint.depthX + minRangeX) / Math.abs(maxRangeX - minRangeX) * maxPixelWidth;

should work

wouterverweirder, 2016-04-14:
depthX and depthY are the positions of a joint in the 2D camera space of the depth camera, in the 0-1 range (0 being left or top, 1 being right or bottom of the 2D depth camera image). But your hand will almost never reach those positions. If you want higher sensitivity / faster movement, you'll need to re-interpolate those values before you multiply with your target window size, something like this:

var minimumX = 0.2;
var maximumX = 0.8;
var x =  window.innerWidth * ((jointPoint.depthX - minimumX) / (maximumX- minimumX));

Loyth, 2016-04-15:
Thanks both of you ! I will test !

closed 2016-04-19 by wouterverweirder