wouterverweirder/kinect2 #31: Issue for MultiSourceReader in a browser

Logs00, 2016-04-15:
Hello,

Sorry in advance for my English ;-)

I have a problem with the MultiSourceReader in browser.

I can't see in the browser the Skeleton and Colors (video) simultaneously.

I can recover Skeleton flux with `kinect.openMultiSourceReader ({frameTypes: Kinect2.FrameType.body});` but I can't retrieve the color flux  with `kinect.openMultiSourceReader ({frameTypes: Kinect2.FrameType.color});`

**Here is my code for the broadcast MultiSourceReader color:**
```
//var Kinect2 = require('../lib/kinect2'),
var Kinect2 = require('kinect2'), //change to 'kinect2' in a project of your own
	express = require('express'),
	app = express(),
	server = require('http').createServer(app),
	io = require('socket.io').listen(server),
	zlib = require('zlib');

var kinect = new Kinect2();

if(kinect.open()) {
    server.listen(8000);
    console.log('Server listening on port 8000');
    console.log('Point your browser to http://localhost:8000');

    app.get('/', function(req, res) {
        res.sendFile(__dirname + '/public/index_multi_video.html');
    });
		app.use(express.static(__dirname + '/public'));

		// compression is used as a factor to resize the image
		// the higher this number, the smaller the image
		// make sure that the width and height (1920 x 1080) are dividable by this number
		// also make sure the canvas size in the html matches the resized size
		var compression = 3;

		var origWidth = 1920;
		var origHeight = 1080;
		var origLength = 4 * origWidth * origHeight;
		var compressedWidth = origWidth / compression;
		var compressedHeight = origHeight / compression;
		var resizedLength = 4 * compressedWidth * compressedHeight;
		//we will send a smaller image (1 / 10th size) over the network
		var resizedBuffer = new Buffer(resizedLength);
		var compressing = false;
    kinect.on('multiSourceFrame', function(data){
				//compress the depth data using zlib
				if(!compressing) {
					compressing = true;
					//data is HD bitmap image, which is a bit too heavy to handle in our browser
					//only send every x pixels over to the browser
					var y2 = 0;
					for(var y = 0; y < origHeight; y+=compression) {
						y2++;
						var x2 = 0;
						for(var x = 0; x < origWidth; x+=compression) {
							var i = 4 * (y * origWidth + x);
							var j = 4 * (y2 * compressedWidth + x2);
							resizedBuffer[j] = data[i];
							resizedBuffer[j+1] = data[i+1];
							resizedBuffer[j+2] = data[i+2];
							resizedBuffer[j+3] = data[i+3];
							x2++;
						}
					}

					zlib.deflate(resizedBuffer, function(err, result){
						if(!err) {
							var buffer = result.toString('base64');
							io.sockets.sockets.forEach(function(socket){
								socket.volatile.emit('multiSourceFrame', buffer);
							});
						}
						compressing = false;
					});
				}
			});

   kinect.openMultiSourceReader({frameTypes: Kinect2.FrameType.color});

}

```
**And my code to display color flux:**

```
<!DOCTYPE html>
<html>
<head>
	<meta charset="utf-8">
	<meta http-equiv="X-UA-Compatible" content="IE=edge">
	<title>Node Kinect2 Client</title>
	<link rel="stylesheet" href="">
</head>
<body>
	<canvas id="bodyCanvas" width="640" height="360"></canvas>
	<script src="/socket.io/socket.io.js"></script>
	<script>
		var socket = io.connect('/');
		var canvas = document.getElementById('bodyCanvas');
		var ctx = canvas.getContext('2d');

		var colorProcessing = false;
		var colorWorkerThread = new Worker("js/colorWorker.js");

		colorWorkerThread.addEventListener("message", function (event) {
			if(event.data.message === 'imageReady') {
                ctx.putImageData(event.data.imageData, 0, 0);
                colorProcessing = false;
			}
		});

		colorWorkerThread.postMessage({
			"message": "setImageData",
			"imageData": ctx.createImageData(canvas.width, canvas.height)
		});

		socket.on('multiSourceFrame', function(imageBuffer){
			if(!colorProcessing) {
				colorProcessing = true;
				colorWorkerThread.postMessage({ "message": "processImageData", "imageBuffer": imageBuffer });
			}
		});
	</script>
</body>
</html>
```

I just can display the video flux with `kinect.openColorReader();` but not with the MultiSourceReader.

Thank in advance for your help :-)

wouterverweirder, 2016-04-15:
Hi,

The color data should be inside data.color.buffer instead of just data:

```javascript
kinect.on('multiSourceFrame', function(data){
    //compress the color data using zlib
    if(!compressing) {
        compressing = true;
        //data.buffer.color is HD bitmap image, which is a bit too heavy to handle in our browser
        //only send every x pixels over to the browser
        var y2 = 0;
        for(var y = 0; y < origHeight; y+=compression) {
            y2++;
            var x2 = 0;
            for(var x = 0; x < origWidth; x+=compression) {
                var i = 4 * (y * origWidth + x);
                var j = 4 * (y2 * compressedWidth + x2);
                resizedBuffer[j] = data.color.buffer[i];
                resizedBuffer[j+1] = data.color.buffer[i+1];
                resizedBuffer[j+2] = data.color.buffer[i+2];
                resizedBuffer[j+3] = data.color.buffer[i+3];
                x2++;
            }
        }

        zlib.deflate(resizedBuffer, function(err, result){
            if(!err) {
                var buffer = result.toString('base64');
                io.sockets.sockets.forEach(function(socket){
                    socket.volatile.emit('multiSourceFrame', buffer);
                });
            }
            compressing = false;
        });
    }
});
```

Logs00, 2016-04-15:
Thank you very very much ;-)
This work perfectly :+1: !

closed 2016-04-15 by wouterverweirder

Logs00, 2016-04-15:
Sorry to repoen this issue, but I have just another question.

To display simoultanesly the skeleton's flux and color's flux I do this : 

broadcast file : 

```
//var Kinect2 = require('../lib/kinect2'),
var Kinect2 = require('kinect2'), //change to 'kinect2' in a project of your own
	express = require('express'),
	app = express(),
	server = require('http').createServer(app),
	io = require('socket.io').listen(server),
	zlib = require('zlib');

var kinect = new Kinect2();

if(kinect.open()) {
    server.listen(8000);
    console.log('Server listening on port 8000');
    console.log('Point your browser to http://localhost:8000');

    app.get('/', function(req, res) {
        res.sendFile(__dirname + '/public/index_multi.html');
    });
		app.use(express.static(__dirname + '/public'));

		// compression is used as a factor to resize the image
		// the higher this number, the smaller the image
		// make sure that the width and height (1920 x 1080) are dividable by this number
		// also make sure the canvas size in the html matches the resized size
		var compression = 2;

		var origWidth = 1920;
		var origHeight = 1080;
		var origLength = 4 * origWidth * origHeight;
		var compressedWidth = origWidth / compression;
		var compressedHeight = origHeight / compression;
		var resizedLength = 4 * compressedWidth * compressedHeight;
		//we will send a smaller image (1 / 10th size) over the network
		var resizedBuffer = new Buffer(resizedLength);
		var compressing = false;
    kinect.on('multiSourceFrame', function(data){
				//compress the depth data using zlib
				if(!compressing) {
					compressing = true;
					//data is HD bitmap image, which is a bit too heavy to handle in our browser
					//only send every x pixels over to the browser
					var y2 = 0;
					for(var y = 0; y < origHeight; y+=compression) {
						y2++;
						var x2 = 0;
						for(var x = 0; x < origWidth; x+=compression) {
							var i = 4 * (y * origWidth + x);
							var j = 4 * (y2 * compressedWidth + x2);
							resizedBuffer[j] = data.color.buffer[i];
							resizedBuffer[j+1] = data.color.buffer[i+1];
							resizedBuffer[j+2] = data.color.buffer[i+2];
							resizedBuffer[j+3] = data.color.buffer[i+3];
							x2++;
						}
					}

					zlib.deflate(resizedBuffer, function(err, result){
						if(!err) {
							var buffer = result.toString('base64');
							io.sockets.sockets.forEach(function(socket){
								socket.volatile.emit('multiSourceFrame', buffer);
							});
						}
						compressing = false;
					});
				}

			},
			function(frame){
					io.sockets.emit('multiSourceFrame', frame);

			});

   kinect.openMultiSourceReader({frameTypes: Kinect2.FrameType.color|Kinect2.FrameType.body});

}

```

And my html file : 

```
<!DOCTYPE html>
<html>
<head>
	<meta charset="utf-8">
	<meta http-equiv="X-UA-Compatible" content="IE=edge">
	<title>Node Kinect2 Client</title>
	<link href="css/style.css" rel="stylesheet" type="text/css" media="all" />
	<link rel="stylesheet" href="">
</head>
<body>
	<canvas id="bodyCanvas" width="960" height="540"></canvas>
	<script src="/socket.io/socket.io.js"></script>
	<script>
		var socket = io.connect('/');
		var canvas = document.getElementById('bodyCanvas');
		var ctx = canvas.getContext('2d');

		var colors = ['#ff0000', '#00ff00', '#0000ff', '#ffff00', '#00ffff', '#ff00ff'];

		// handstate circle size
		var HANDSIZE = 20;

		// closed hand state color
		var HANDCLOSEDCOLOR = "red";

		// open hand state color
		var HANDOPENCOLOR = "green";

		// lasso hand state color
		var HANDLASSOCOLOR = "blue";

		function updateHandState(handState, jointPoint) {
			switch (handState) {
				case 3:
					drawHand(jointPoint, HANDCLOSEDCOLOR);
				break;

				case 2:
					drawHand(jointPoint, HANDOPENCOLOR);
				break;

				case 4:
					drawHand(jointPoint, HANDLASSOCOLOR);
				break;
			}
		}

		function drawHand(jointPoint, handColor) {
			// draw semi transparent hand cicles
			ctx.globalAlpha = 0.75;
			ctx.beginPath();
			ctx.fillStyle = handColor;
			ctx.arc(jointPoint.depthX * 960, jointPoint.depthY * 540, HANDSIZE, 0, Math.PI * 2, true);
			ctx.fill();
			ctx.closePath();
			ctx.globalAlpha = 1;
		}

		var colorProcessing = false;
		var colorWorkerThread = new Worker("js/colorWorker.js");

		colorWorkerThread.addEventListener("message", function (event) {
			if(event.data.message === 'imageReady') {
                ctx.putImageData(event.data.imageData, 0, 0);
                colorProcessing = false;
			}
		});

		colorWorkerThread.postMessage({
			"message": "setImageData",
			"imageData": ctx.createImageData(canvas.width, canvas.height)
		});

		socket.on('multiSourceFrame', function(imageBuffer){
			if(!colorProcessing) {
				colorProcessing = true;
				colorWorkerThread.postMessage({ "message": "processImageData", "imageBuffer": imageBuffer });
			}

		},
		function(frame){
			ctx.clearRect(0, 0, canvas.width, canvas.height);
			var index = 0;
			frame.body.bodies.forEach(function(body){
				if(body.tracked) {
					for(var jointType in body.joints) {
						var joint = body.joints[jointType];
						ctx.fillStyle = colors[index];
						ctx.fillRect(joint.depthX * 960, joint.depthY * 540, 10, 10);
					}
					//draw hand states
					updateHandState(body.leftHandState, body.joints[7]);
					updateHandState(body.rightHandState, body.joints[11]);
					index++;
				}
			});

		});
	</script>
</body>
</html>
```

With this I just have the video's flux but not the skeleton one.

An idea where i make a mistake?

Thank you very much in advance for your help :-)