wouterverweirder/kinect2 #32: Draw Hand, css position, transition not fluid

Loyth, 2016-04-18:
Hello every one, I have a problem with drawing the hand. I would like the movements of the hand, on the browser, become fluid and not jerky. 

Do you have a solution ?
Thanks.

wouterverweirder, 2016-04-19:
You could add some easing logic. e.g something like this:

```javascript
if (!handStable) {
    if (!handOutOfBounds_width) {
        var currentX = parseFloat(hand.css('left').replace('px', ''));
        hand.css('left', currentX + (left_hand_x - currentX) * 0.1);
    }

    if (!handOutOfBounds_height) {
        var currentY = parseFloat(hand.css('top').replace('px', ''));
        hand.css('top', currentY + (top_hand_y - currentY) * 0.1);
    }
}
```

Loyth, 2016-04-19:
Thanks you very much ! However, i change the sensibility to 0.25 because it was very slow.

closed 2016-04-19 by wouterverweirder