wouterverweirder/kinect2 #33: Node.js server High CPU Usage

Loyth, 2016-04-20:
Hello everybody, this is my Kinect Node.js server

`var Kinect2 = require('kinect2'),
    express = require('express'),
    app = express(),
    server = require('http').createServer(app),
    io = require('socket.io').listen(server),
    zlib = require('zlib');

var kinect = new Kinect2();

if (kinect.open()) {
    server.listen(8000);
    console.log('Serveur ecoute sur le port 8000');
    console.log('Lancez votre navigateur sur http://localhost:8000');

    app.use(express.static(__dirname + '/public2'));
    
    kinect.on('bodyFrame', function (bodyFrame) {
        io.sockets.emit('bodyFrame', bodyFrame);
    });

    kinect.openBodyReader();
}`

When I go to the Task Manager, Node.js use 26% of CPU.

Do you have a solution to decrease that ?
Thanks.

wouterverweirder, 2016-04-21:
Do you see a significant difference with launching some of the standard kinect SDK samples? Kinect 2 does demand a lot from your system.

Loyth, 2016-04-21:
Yes, there is a significant difference. When I launch Controls Basics SDK Sample, the KinectService.exe use 10% of CPU and the app ControlsBasicsWPF.exe use 15%.

When I launch my App on the browser, the KinectService.exe use already 10% of CPU but the Node.js Server use 60% of CPU (because I stream BodyReader and DepthReader).

wouterverweirder, 2016-04-21:
What extra processing do you do inside of your nodejs app? Sending image streams (such as depth, rgb) will demand extra CPU resources as node needs to compress / serialize that data. If you want to handle image streams and want high performance, I suggest using it in an electron app instead of serializing it over a socket to the browser.

Loyth, 2016-04-21:
I only launch bodyReader and depthReader in my nodejs app.

I try to build electron app but I have some errors with

`npm install electron-kinect2`

closed 2016-05-10 by wouterverweirder