wouterverweirder/kinect2 #38: Joints smoothing

gabrielstuff, 2016-05-14:
Hi there,

We are working with your tool which works really great. We were wondering if anyone got interest in JOINT smoothing and constrains.

We found this algorithms : https://github.com/KinectEx/KinectEx/tree/master/src/KinectEx/Smoothing

Anyone already tried ?

wouterverweirder, 2016-05-14:
Looks interesting - might be something for an additional / external js library which would process the feed from kinect2.js according to those algorithms.

gabrielstuff, 2016-05-14:
Well, i'm wondering about the perf... This should be done natively i think.
We implemented simple filters in JS. But more complex ones could eat some
CPUs and so be ported in C++ or .Net
Then in Kinect you can simply choose a filter when you init the capture.

wouterverweirder, 2016-09-12:
Will not implement this as a core functionality of this library. This would require keeping snapshots of multiple frames in C++ memory, which is easier to manage in JS, without too much overhead. I don't think it's performance critical to do this in C++.

closed 2016-09-12 by wouterverweirder

gabrielstuff, 2016-09-12:
Hey !
We implemented some in JS, but far from the efficiency found in C++ libs. Our methods needs lots of work . Nonetheless thanks for keeping us updated.