wouterverweirder/kinect2 #39: fatal error C1083: Cannot open include file: 'Kinect.h': No such file or directory [C:\path\to\local\kinect2\build

nikolakanacki, 2016-09-05:


nikolakanacki, 2016-09-05:
Any way anyone else has the same issue? Trying to build for react for last like ~8 hours. Not a windows expert, but this is where it fails. I have installed the official Kinect 2.0 SDK and have a C++ Visual Studio Express installed as well. This happens when using the script provided in the package `tools/electronbuild.js`

Any help would be appreciated.

closed 2016-09-05 by nikolakanacki

marklundin, 2016-09-15:
@nikolakanacki What was your solution for this? I'm having exactly the same issue.

nikolakanacki, 2016-09-15:
So it turns out that, when up for three days trying to meet a deadline, one can forget which of the testing boxes have the Official SDK set up on them.

My problem was that the one I was trying (failing) to compile the app on didn't have one, so it was reasonable for the compiler to error with `Cannot open Include file: `'Kinect.h'`.

Installing it solved the problem.

The official SDK can be found here: https://www.microsoft.com/en-us/download/details.aspx?id=44561

nikolakanacki, 2016-09-15:
I just closed the issue, but thinking back - I should've posted the "solution" for all the other caffeine burners.

marklundin, 2016-09-15:
Ha! Yes, that does look like the issue. Shame they don't have Kinect SDK on npm ;) 

Thanks for your help

nikolakanacki, 2016-09-15:
Someday! :D glad I could help, any time

marklundin, 2016-09-15:
Hmm, seems that after installing the SDK and rebooting I'm still getting the same issue;

`..\src\kinect2.cc(4): fatal error C1083: Cannot open include file: 'Kinect.h': No such file or directory`

Not sure what could be the issue, but it seems it still can't resolve the `Kinect.h` header

nikolakanacki, 2016-09-15:
Do you have `C++ Visual Studio Express` installed as well? Read somewhere it was a must.

nikolakanacki, 2016-09-15:
I'm just trowing things out there, not a windows guy but did manage to make it work.

marklundin, 2016-09-15:
Yep, I installed as per https://github.com/nodejs/node-gyp#installation

nikolakanacki, 2016-09-15:
What about the OS?

marklundin, 2016-09-15:
Yep, Windows 10 Home