wouterverweirder/kinect2 #40: Matting based on depth images of the Kinect

cnwander, 2016-09-23:
is there has a method like "AccessUnderlyingBuffer" to get the depth of colorpixel? thanks a lot

wouterverweirder, 2016-09-23:
Could you elaborate what you're trying to achieve? You can get a combination of data (depth, color, ...) with `kinect.openMultiSourceReader({frameTypes: Kinect2.FrameType.rawDepth | Kinect2.FrameType.color})` and listening for multiSourceFrame events?

cnwander, 2016-09-26:
thanks. 
i'm trying to achieve is body image matting, and change the backgroud, like this:
[image]
[image]

newmovieclip, 2016-09-26:
Aha, https://github.com/wouterverweirder/kinect2/tree/master/examples/greenkey-electron is an example doing just that :-)

closed 2016-09-26 by wouterverweirder