wouterverweirder/kinect2 #44: Save colorData to File

mariovde, 2017-02-08:
Hi

This is not so much an issue but:
How would you write colorFrame-data (<Buffer ...>) to an actual file ?

regards,

Mario

wouterverweirder, 2017-02-11:
Well, the pixel buffer you get are raw pixel values (bitmap). You should be able to save that data as a .bmp.

If you need compression, you could take a look at examples/color-feed-browser-lwip/index.js - in that example I'm compressing the bitmap as a jpeg (using lwip), before I'm sending it over a socket connection.

mariovde, 2017-02-13:
When using this code, it says the file is save, but the bmp-file won't open.
It's 8mb large though...

var kinect = new Kinect2();
  if(kinect.open()){
    kinect.on('colorFrame', function(newPixelData){
      kinect.close();
      console.log(newPixelData);
      fs.writeFile("C:/test.bmp", newPixelData,  "binary",function(err) {
        if(err) {
          console.log(err);
        } else {
          console.log("The file was saved!");
        }
      });

    });
    kinect.openColorReader();
  }

ACPrice, 2018-05-14:
Hey @mariovde I know this is an old thread, but the `newPixelData` you're using above is only the actual data for the pixel values. To save a bitmap that can be opened, the `.bmp` you save needs to also have additional data _before_ those pixel values. That data constitutes the bitmap file "header". 

Check out how this repo adds headers to the bitmap file:

https://github.com/shaozilee/bmp-js/blob/master/lib/encoder.js

That code won't work out of the box for raw Kinect data, but it's close, and a good example regardless.

hzjdandan, 2018-10-11:
@mariovde   did you solve the problem?

closed 2018-10-11 by mariovde

hzjdandan, 2018-10-11:
> @hzjdandan yes, like described above by ACPrice.

Is it convenient to provide code?    Thank you very much!