wouterverweirder/kinect2 #47: Ask about how to use handstate

wildanihadi, 2017-05-10:
I want to ask how to access handstate? Sorry if I ask like that because I'm still beginner in learning programming.
As an example I am trying to display text like this code 

`var lefthand = bodyFrame.bodies[i].joints[Kinect2.JointType.handLeft].cameraY;
   var head = bodyFrame.bodies[i].joints[Kinect2.JointType.head].cameraY;

	if (lefthand>head){
		console.log("Your hand over your head")
	}`
Then how to display text if both hands are closed or open?
I am trying to see what the contents of the handstate with this code but that result in the command prompt is [Object Object]

`console.log (bodyFrame.bodies[i].joints[Kinect2.JointType.HandState]);`

And I have no idea how to access it to display text. I hope someone can help me. thanks.

wouterverweirder, 2017-05-10:
You can read the handstate on a body:

`bodyFrame.bodies[i].leftHandState`

this will be one of the following values:

- Kinect2.HandState.closed
- Kinect2.HandState.open
- Kinect2.HandState.lasso

You can find a handstate example in examples/skeletons-electron/index.html

wildanihadi, 2017-05-10:
Thanks for your answer. I really appreciate that.

closed 2017-05-10 by wouterverweirder