wouterverweirder/kinect2 #49: joint tracked info

bang, 2017-08-23:
Hi!

First of all, excelent job! You made my life easier when I try to started projects using Kinect 2. Thank you!

However, I'm trying to calculate the height from skeleton and this is almost impossible without trackingState info of the joints(not the body, of each joint). I'm trying to follow this tutorial (http://pterneas.com/2012/06/08/kinect-for-windows-find-user-height-accurately/), where the results appears to be very accurate. But it's all written in C# what I want to avoid it if I can. Do you know another way to calculate height without using trackingState? I´ve come up with of using  estimated values based on calculation including circunference and diameter of skull. But this is very difficult for me to get from Kinect. I've tried to check on C++ library in your repository if the information was there or if it's possible to get this information from it or even from the 'Kinect.h'. Well, I failed in both.

Can you help me/guide me, please?

Thanks!

wouterverweirder, 2017-08-26:
Take a look at the floor-height example: here I calculate the distance from a joint to the floor plane:

https://github.com/wouterverweirder/kinect2/blob/master/examples/floor-height-joint-electron/js/app.js

mayilook, 2017-12-30:
Hello,
bang was asking about joint tracking state (https://msdn.microsoft.com/en-us/library/microsoft.kinect.jointtrackingstate.aspx), it's show by kinect lib, is it possible to get it via your module?

wouterverweirder, 2018-08-09:
Just pushed an update. Each joint has a new property trackingState:

```
joint.trackingState
```

Which can be any of the following values:

```
Kinect2.TrackingState.notTracked (0)
Kinect2.TrackingState.inferred (1)
Kinect2.TrackingState.tracked (2)
```

closed 2018-08-09 by wouterverweirder