wouterverweirder/kinect2 #53: Tried to download(404)

shahho, 2019-05-05:
node-pre-gyp ERR! Tried to download(404): https://github.com/wouterverweirder/kinect2/releases/download/0.1.6/node-v67-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for kinect2@0.1.6 and node@11.0.0 (node-v67 ABI, unknown) (falling back to source compile with node-gyp)
node-pre-gyp ERR! Tried to download(undefined): https://github.com/wouterverweirder/kinect2/releases/download/0.1.6/node-v67-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for kinect2@0.1.6 and node@11.0.0 (node-v67 ABI, unknown) (falling back to source compile with node-gyp)

why?

cecilemuller, 2019-05-05:
Looks like it's because it's a newer version of Node than what existed back in 2018 when this package was last modified, it only goes up to `v64` on the release page: https://github.com/wouterverweirder/kinect2/releases/tag/0.1.6

wouterverweirder, 2019-05-05:
I've published a new release (v0.1.7), with support for the latest node versions - could you test if it works now?

closed 2019-05-09 by wouterverweirder

cecilemuller, 2019-05-09:
I don't know for shahho, but I tried 0.1.7 just now on latest Node and the installation worked fine, good job 👍