wouterverweirder/kinect2 #55: Support for new version of Kinect

shreyapandit, 2019-09-29:
Hi there,

Thanks for the great work on your package 🥇 

Microsoft recently announced the new Kinect sensor:
https://azure.microsoft.com/en-us/services/kinect-dk/

Would this package work with the new sensor as well?

Thanks!

wouterverweirder, 2019-10-01:
Hey - Would love to, trying to get my hands on a sensor as it's not available yet in Europe...

shreyapandit, 2019-10-05:
Thanks for your response! I am based out of US and will get a sensor next week. Would love to constribute if you could help me get started on what would be first steps in that regard. Cheers!

wouterverweirder, 2019-11-18:
Got my Azure Kinect last week and started work on an extension: https://github.com/wouterverweirder/kinect-azure

closed 2020-02-07 by wouterverweirder