wouterverweirder/kinect2 #57: Package no longer builds due to deprecated functions

Chrisbright10, 2020-01-23:
Upon trying to build this package I ran into the following output. It seems that deprecated functions and changes in the v8 code have taken effect and must be fixed in the project before it can build again. Would you mind updating the functions in the package?

`{MY_DIR}\.electron-gyp\7.1.10\include\node\v8.h(8513): warning C4996: 'v8::MicrotasksCompletedCallback': was declared deprecated (compiling source file ..\src\kinect2.cc) [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {MY_DIR}\.electron-gyp\7.1.10\include\node\v8.h(6918): note: see declaration of 'v8::MicrotasksCompletedCallback' (compiling source file ..\src\kinect2.cc)
{MY_DIR}\.electron-gyp\7.1.10\include\node\v8.h(8522): warning C4996: 'v8::MicrotasksCompletedCallback': was declared deprecated (compiling source file ..\src\kinect2.cc) [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {MY_DIR}\.electron-gyp\7.1.10\include\node\v8.h(6918): note: see declaration of 'v8::MicrotasksCompletedCallback' (compiling source file ..\src\kinect2.cc)
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(205): warning C4996: 'Nan::Callback::Call': was declared deprecated [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {PROJECT_DIR}\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(365): warning C4996: 'Nan::Callback::Call': was declared deprecated [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {PROJECT_DIR}\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(536): warning C4996: 'Nan::Callback::Call': was declared deprecated [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {PROJECT_DIR}\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(708): warning C4996: 'Nan::Callback::Call': was declared deprecated [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {PROJECT_DIR}\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(877): warning C4996: 'Nan::Callback::Call': was declared deprecated [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {PROJECT_DIR}\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1106): warning C4996: 'Nan::Callback::Call': was declared deprecated [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {PROJECT_DIR}\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1411): error C2661: 'v8::Object::Get': no overloaded function takes 1 arguments [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1430): warning C4996: 'Nan::Callback::Call': was declared deprecated [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {PROJECT_DIR}\node_modules\nan\nan.h(1740): note: see declaration of 'Nan::Callback::Call'
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1751): error C2661: 'v8::Object::Get': no overloaded function takes 1 arguments [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1751): error C2059: syntax error: ')' [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1752): error C2661: 'v8::Object::Get': no overloaded function takes 1 arguments [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1752): error C2059: syntax error: ')' [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1752): error C2143: syntax error: missing '(' before ')' [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1858): error C2661: 'v8::Object::Get': no overloaded function takes 1 arguments [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1858): error C2059: syntax error: ')' [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
{PROJECT_DIR}\node_modules\kinect2\src\kinect2.cc(1858): error C2143: syntax error: missing '(' before ')' [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
{MY_DIR}\.electron-gyp\7.1.10\include\node\v8.h(8513): warning C4996: 'v8::MicrotasksCompletedCallback': was declared deprecated (compiling source file ..\src\kinect2.cc) [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {MY_DIR}\.electron-gyp\7.1.10\include\node\v8.h(6918): note: see declaration of 'v8::MicrotasksCompletedCallback' (compiling source file ..\src\kinect2.cc)
{MY_DIR}\.electron-gyp\7.1.10\include\node\v8.h(8522): warning C4996: 'v8::MicrotasksCompletedCallback': was declared deprecated (compiling source file ..\src\kinect2.cc) [{PROJECT_DIR}\node_modules\kinect2\build\kinect2.vcxproj]
  {MY_DIR}\.electron-gyp\7.1.10\include\node\v8.h(6918): note: see declaration of 'v8::MicrotasksCompletedCallback' (compiling source file ..\src\kinect2.cc)
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module={PROJECT_DIR}\node_modules\kinect2\build\kinect2\v0.1.7\Release\electron-v7.1-win32-x64\kinect2.node --module_name=kinect2 --module_path={PROJECT_DIR}\node_modules\kinect2\build\kinect2\v0.1.7\Release\electron-v7.1-win32-x64' (1)`

wouterverweirder, 2020-04-12:
No promises, but I'll look into how much work it would be to port it over to the architecture I'm using in my kinect-azure library - making it compatible with modern node versions & use it directly in electron without recompiling...

wouterverweirder, 2020-04-14:
Just released version 0.2.0 of this library, which is a complete rewrite on the native C side, making it compatible with modern node versions. Additional benefit is that no rebuilds are needed to use it in electron apps.

closed 2020-04-14 by wouterverweirder