wouterverweirder/kinect2 #59: Starting and stopping cameras crashes my code

Chrisbright10, 2020-05-12:
https://github.com/wouterverweirder/kinect-azure/issues/6

wouterverweirder, 2020-05-12:
Ah, thanks for catching that one. Unfortunately, I don't have access to my windows machine right now, as I've lent it to a colleague of mine who had computer problems. So, it'll have to wait until I have my machine back 😅

Chrisbright10, 2020-05-12:
No worries, but thank you for being so willing to update it! You're a champ!

wouterverweirder, 2020-05-23:
Just released v0.2.1 - which fixes this issue.
You can now await the close functions or pass in a callback, as illustrated in the demo below. It works for all close functions and for the generic kinect.close function as well:

https://github.com/wouterverweirder/kinect2/blob/1879a00d9bc0a81458b39e4d5f2f0712f84615a0/examples/node/restart-multiple-times-color.js#L1-L33

commit 1879a00d (2020-05-23): wait for threadsafe function to finish - fixes #59
closed 2020-05-23 by wouterverweirder