wouterverweirder/kinect2 #62: Video stream reader issue

hugodecasta, 2020-10-16:
Opening a video stream (Depth, Infrared, Color) rises a `Fatal error` issue:

here is the code used
```js
const Kinect2 = require('kinect2');
const kinect = new Kinect2();

const main = async () => {
    if (kinect.open()) {
        console.log("Kinect Opened");
        kinect.openInfraredReader();
    }
};

main();
```

OS:
 * Windows 10 - 64bits
 * node v14.12.0

Error produced:
```
#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 000000FE8B6FF180
 1: 00007FF6424513CF napi_wrap+109135
 2: 00007FF642384C9F std::basic_ostream<char,std::char_traits<char> >::operator<<+56591
 3: 00007FF642FC2A52 V8_Fatal+162
 4: 00007FF642A6508D v8::internal::BackingStore::Reallocate+653
 5: 00007FF642CAC509 v8::ArrayBuffer::GetBackingStore+137
 6: 00007FF642433699 napi_get_typedarray_info+393
 7: 00007FFED6ED626E
 8: 00007FFED6EE0DB5
 9: 00007FFED6ED9775
10: 00007FFED6EE3052
11: 00007FFED6EDB123
12: 00007FF64241DC5E node_module_register+4030
13: 00007FF64249A6D8 uv_idle_init+120
14: 00007FF6424A4F54 uv_run+212
15: 00007FF6423AFEE5 v8::internal::OrderedHashTable<v8::internal::OrderedHashMap,2>::NumberOfBucketsOffset+9477
16: 00007FF642424AD7 node::Start+311
17: 00007FF64228678C RC4_options+339596
18: 00007FF64322A65C v8::internal::compiler::RepresentationChanger::Uint32OverflowOperatorFor+152748
19: 00007FFF2D447974 BaseThreadInitThunk+20
20: 00007FFF2DB5A271 RtlUserThreadStart+33
```

All these methods fail to open and rises this error
 * openColorReader 
 * openDepthReader 
 * openRawDepthReader 
 * openInfraredReader
 * openLongExposureInfraredReader

The only working opener is
 * openBodyReader

wouterverweirder, 2020-10-17:
Thanks for the report - I currently do not have access to my windows machine, but will investigate when I get it back (will be a couple of weeks though).

In the meantime: could you try on a previous nodejs / electron version and see if you get the same behaviour?

wouterverweirder, 2021-03-09:
Sorry for the long time it took - do you still have this issue with the latest version of this module? (v0.2.2)

hugodecasta, 2021-03-19:
i'd need to try it again, coming back when i've done it ;)

quez-fun, 2021-07-15:
> Sorry for the long time it took - do you still have this issue with the latest version of this module? (v0.2.2)

Yes, the issue persists with v0.2.2 (run with the latest nodejs). 

btw I wonder why Electron handles video stream demos gracefully, whereas nodejs + express + browser html cannot.

quez-fun, 2021-07-15:
> i'd need to try it again, coming back when i've done it ;)

actually I just got this solved. 

wouterverweirder is right - try a previous nodejs version. In my case I tried v8.17.0 as it is the earliest nodejs that supports napi 4 (which is required for this module to work).

hugodecasta, 2021-07-15:
> > i'd need to try it again, coming back when i've done it ;)
> 
> actually I just got this solved.
> 
> wouterverweirder is right - try a previous nodejs version. In my case I tried v8.17.0 as it is the earliest nodejs that supports napi 4 (which is required for this module to work).

Well, i understand but isn't it kind of a problem to have to downgrade your node to run a program (isn't it more standard to have the program updated for the newest versions) ?

quez-fun, 2021-07-16:
> Well, i understand but isn't it kind of a problem to have to downgrade your node to run a program (isn't it more standard to have the program updated for the newest versions) ?

Stick to the electron demos if you want to use the latest nodejs. Somehow running openColorReader/openDepthReader/etc inside the electron framework does not trigger the same issue as in bare nodejs -- I wonder why, though.

You might also consider running multiple nodejs versions on your machine.

wouterverweirder, 2021-07-16:
I've just pushed an update to the way prebuild binaries are handled - could you try v0.2.3 and see if the error persists? The whole idea of napi is to be future compatible. I've tested on node v14 as well, and it is working on my end. If not: could you post your CLI log of what happens when you npm install?

quez-fun, 2021-07-16:
> I've just pushed an update to the way prebuild binaries are handled - could you try v0.2.3 and see if the error persists? The whole idea of napi is to be future compatible. I've tested on node v14 [...]

hi there, thx 4 the update!

I've just tried v0.2.3 with two latest versions of nodejs (v14 and v16). The 'test.js' file you've just included works fine, until commenting out the following codes:

//if (hasAllInfo) {
      //console.log("close multi source reader");
      //kinect.closeMultiSourceReader(() => {
        //console.log("stopped multi source reader");
        //kinect.close();
      //});
    //}

Then 'fatal error' comes up again, which is like 

#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 000000FC1F56ECA0
 1: 00007FF698FF481F napi_wrap+110783
 2: 00007FF698F268DF std::basic_ostream<char,std::char_traits<char> >::operator<<+57887
 3: 00007FF699B88F82 V8_Fatal+162
 4: 00007FF69961158D v8::internal::BackingStore::Reallocate+653
 5: 00007FF699858EE9 v8::ArrayBuffer::GetBackingStore+137
 6: 00007FF698FD60D9 napi_get_typedarray_info+393
 7: 00007FF8A4E1FCDE
 8: 00007FF8A4E1E3EF
 9: 00007FF8A4E20E8D
10: 00007FF8A4E13BCF
11: 00007FF698FBFE3E node_module_register+4638
12: 00007FF698FBF6D1 node_module_register+2737
13: 00007FF69905221B uv_async_send+331
14: 00007FF6990519AC uv_loop_init+1292
15: 00007FF699051B4A uv_run+202
16: 00007FF698F52BF5 v8::internal::OrderedHashTable<v8::internal::OrderedHashMap,2>::NumberOfBucketsOffset+9589
17: 00007FF698FC70F7 node::Start+311
18: 00007FF698E2679C RC4_options+339612
19: 00007FF699DF0DE8 v8::internal::compiler::RepresentationChanger::Uint32OverflowOperatorFor+154568
20: 00007FF904907034 BaseThreadInitThunk+20
21: 00007FF9064E2651 RtlUserThreadStart+33

in node v14.17.3, and 

#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 000000451D2FEEC0
 1: 00007FF798A9A73F v8::internal::CodeObjectRegistry::~CodeObjectRegistry+111711
 2: 00007FF7989BAC7F v8::CFunctionInfo::HasOptions+7039
 3: 00007FF7996E1E12 V8_Fatal+162
 4: 00007FF79913483D v8::internal::BackingStore::Reallocate+653
 5: 00007FF7993A8C09 v8::ArrayBuffer::GetBackingStore+137
 6: 00007FF798A6EDB9 napi_get_typedarray_info+393
 7: 00007FF891F82421
 8: 00007FF891F80C92
 9: 00007FF891F83A9E
10: 00007FF891F74238
11: 00007FF798A58D7E node_module_register+4798
12: 00007FF798A58611 node_module_register+2897
13: 00007FF798AFA98B uv_async_send+331
14: 00007FF798AFA11C uv_loop_init+1292
15: 00007FF798AFA2BA uv_run+202
16: 00007FF798AC9565 node::SpinEventLoop+309
17: 00007FF7989E5023 EVP_CIPHER_CTX_buf_noconst+53651
18: 00007FF798A5FF1D node::Start+221
19: 00007FF79888844C RC4_options+346956
20: 00007FF799940398 v8::internal::compiler::RepresentationChanger::Uint32OverflowOperatorFor+14456
21: 00007FF904907034 BaseThreadInitThunk+20
22: 00007FF9064E2651 RtlUserThreadStart+33

in node v16.4.2. 

These error reports are exactly the same as in kinect2 v0.2.2 except for the 16 digits in '#FailureMessage Object: XXXX XXXX XXXX XXXX' which always differ in each test.

**The key issue is that both kinect2 v0.2.2 and v0.2.3 work fine to read ONE SINGLE FRAME**, be it depth frame or color frame etc. I tested for many times and found that v0.2.2 can read and log an entire color frame (8294400 bytes long) or depth frame (217088 bytes), and calmly exit (by calling process.exit() after reading one frame) without reporting an error.

It is when programmed to read **more than two frames** that error reports occur, not to mention continuously read.

Interestingly, the same codes that produce fatal errors in bare nodejs work perfectly in electron.

quez-fun, 2021-07-16:
> I've just pushed an update to the way prebuild binaries are handled - could you try v0.2.3 and see if the error persists? The whole idea of napi is to be future compatible. I've tested on node v14 [...]

hi there, thx 4 the update!

I've just tried v0.2.3 with two latest versions of nodejs (v14 and v16). The 'test.js' file you've just included works fine, until commenting out the following codes:

```
//if (hasAllInfo) {
//console.log("close multi source reader");
//kinect.closeMultiSourceReader(() => {
//console.log("stopped multi source reader");
//kinect.close();
//});
//}
```
Then 'fatal error' comes up again, which is like

```
#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 000000FC1F56ECA0
1: 00007FF698FF481F napi_wrap+110783
2: 00007FF698F268DF std::basic_ostream<char,std::char_traits >::operator<<+57887
3: 00007FF699B88F82 V8_Fatal+162
4: 00007FF69961158D v8::internal::BackingStore::Reallocate+653
5: 00007FF699858EE9 v8::ArrayBuffer::GetBackingStore+137
6: 00007FF698FD60D9 napi_get_typedarray_info+393
7: 00007FF8A4E1FCDE
8: 00007FF8A4E1E3EF
9: 00007FF8A4E20E8D
10: 00007FF8A4E13BCF
11: 00007FF698FBFE3E node_module_register+4638
12: 00007FF698FBF6D1 node_module_register+2737
13: 00007FF69905221B uv_async_send+331
14: 00007FF6990519AC uv_loop_init+1292
15: 00007FF699051B4A uv_run+202
16: 00007FF698F52BF5 v8::internal::OrderedHashTablev8::internal::OrderedHashMap,2::NumberOfBucketsOffset+9589
17: 00007FF698FC70F7 node::Start+311
18: 00007FF698E2679C RC4_options+339612
19: 00007FF699DF0DE8 v8::internal::compiler::RepresentationChanger::Uint32OverflowOperatorFor+154568
20: 00007FF904907034 BaseThreadInitThunk+20
21: 00007FF9064E2651 RtlUserThreadStart+33
```

in node v14.17.3, and

```
#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 000000451D2FEEC0
1: 00007FF798A9A73F v8::internal::CodeObjectRegistry::~CodeObjectRegistry+111711
2: 00007FF7989BAC7F v8::CFunctionInfo::HasOptions+7039
3: 00007FF7996E1E12 V8_Fatal+162
4: 00007FF79913483D v8::internal::BackingStore::Reallocate+653
5: 00007FF7993A8C09 v8::ArrayBuffer::GetBackingStore+137
6: 00007FF798A6EDB9 napi_get_typedarray_info+393
7: 00007FF891F82421
8: 00007FF891F80C92
9: 00007FF891F83A9E
10: 00007FF891F74238
11: 00007FF798A58D7E node_module_register+4798
12: 00007FF798A58611 node_module_register+2897
13: 00007FF798AFA98B uv_async_send+331
14: 00007FF798AFA11C uv_loop_init+1292
15: 00007FF798AFA2BA uv_run+202
16: 00007FF798AC9565 node::SpinEventLoop+309
17: 00007FF7989E5023 EVP_CIPHER_CTX_buf_noconst+53651
18: 00007FF798A5FF1D node::Start+221
19: 00007FF79888844C RC4_options+346956
20: 00007FF799940398 v8::internal::compiler::RepresentationChanger::Uint32OverflowOperatorFor+14456
21: 00007FF904907034 BaseThreadInitThunk+20
22: 00007FF9064E2651 RtlUserThreadStart+33
```

in node v16.4.2.

These error reports are exactly the same as in kinect2 v0.2.2 except for the 16 digits in '#FailureMessage Object: XXXX XXXX XXXX XXXX' which always differ in each test.

The key issue is that both kinect2 v0.2.2 and v0.2.3 work fine to read ONE SINGLE FRAME, be it depth frame or color frame etc. I tested for many times and found that v0.2.2 can read and log an entire color frame (8294400 bytes long) or depth frame (217088 bytes), and calmly exit (by calling process.exit() after reading one frame) without reporting an error.

It is when programmed to read more than two frames that error reports occur, not to mention continuously read.

Interestingly, the same codes that produce fatal errors in bare nodejs work perfectly in electron.

wouterverweirder, 2021-07-31:
Looks like there were some changes in the way buffers are handled in node 14 (https://github.com/nodejs/node/issues/32463)

I've made some changes and pushed a new version 0.2.4: https://github.com/wouterverweirder/kinect2/releases/tag/v0.2.4 - hopefully this fixes the issues on your end

quez-fun, 2021-09-08:
> Looks like there were some changes in the way buffers are handled in node 14 ([nodejs/node#32463](https://github.com/nodejs/node/issues/32463))
> 
> I've made some changes and pushed a new version 0.2.4: https://github.com/wouterverweirder/kinect2/releases/tag/v0.2.4 - hopefully this fixes the issues on your end

hi wouterverweirder, I just tested your new package on node-14.17.6 and node-16.9.0, both worked perfectly, with new console lines:

```
[kinect2.cc] m_enabledFrameTypes: 40
[kinect2.cc] m_enabledFrameSourceTypes: 40
(node:5572) [DEP0066] DeprecationWarning: OutgoingMessage.prototype._headers is deprecated
(Use `node --trace-deprecation ...` to show where the warning was created)
```

(in the case of node-16.9.0, it's node:13876 instead of node:5572)

Your project is of great value to me, and I would like to donate 10 euros as a small token of gratitude. However, I failed to make a donation because of country restriction (I'm currently in China). Please tell me if there is any alternative donation methods that are available; or I might ask if any of my friends in EU would like to help.

Thanks again for the excellent project!

quez-fun, 2021-09-08:
> Looks like there were some changes in the way buffers are handled in node 14 ([nodejs/node#32463](https://github.com/nodejs/node/issues/32463))
>
> I've made some changes and pushed a new version [...]

hi a friend of a friend of mine has just made a pay-pal transfer (email: For**88@live.com). thx again for the excellent project, I've learnt a great deal from it.

wouterverweirder, 2021-09-08:
Glad to hear this issue is fixed & thanks a lot for the donation! Appreciated :-)

closed 2021-09-08 by wouterverweirder