wouterverweirder/kinect2 #63: Issue when installing

spotandjake, 2020-10-16:
when i go to install the module with npm i get the error "floor" is not a member of "std"

gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:314:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:276:12)
gyp ERR! System Windows_NT 10.0.19041
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\Spota\node_modules\kinect2
gyp ERR! node -v v14.13.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok

wouterverweirder, 2020-10-17:
Thanks for the report - I currently do not have access to my windows machine, but will investigate when I get it back (will be a couple of weeks though).

In the meantime: could you try on a previous nodejs / electron version and see if you get the same behaviour?

spotandjake, 2020-10-18:
i tried it on the 3 versions of node below and all of them errored with the same issue 
12.19.0
10.22.1
 8.17.0

wouterverweirder, 2020-10-18:
Looks like it's probably related to VS 2019 - referencing to https://github.com/microsoft/terminal/pull/449/files for future investigation

spotandjake, 2020-10-18:
sad is it good with vs 2019, it is unfortunate as i hate installing visual studio but thks, will try to install visual studio 2017 for now . better awnser than it is just an issue with my computer for me anyways lol. thanks for a great libary cant wait to get it working.

spotandjake, 2020-10-18:
Deffinetly an issue with visual studio 2019 i uninstalled it and it worked instantly

wouterverweirder, 2021-03-09:
I've released a new version (v0.2.2), which now uses precompiled binaries - this hopefully fixes these issues.

closed 2021-03-09 by wouterverweirder

spotandjake, 2021-03-09:
Massively appreciated thank you